PRBonn/depth_clustering #1: *** Error in `./show_objects_moosman': realloc(): invalid pointer: 0x00007fae44ebb820 ***

suijingfeng, 2016-12-20:
*** Error in `./show_objects_moosman': realloc(): invalid pointer: 0x00007fae44ebb820 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7fae480ae7e5]
/lib/x86_64-linux-gnu/libc.so.6(realloc+0x348)[0x7fae480bae58]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN9QListData7reallocEi+0x1f)[0x7fae433e09cf]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN9QListData6appendEi+0x81)[0x7fae433e0aa1]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x1d6d78)[0x7fae434acd78]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_Z21qRegisterResourceDataiPKhS0_S0_+0x2e6)[0x7fae434a8b16]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x7bcc3)[0x7fae43351cc3]
/lib64/ld-linux-x86-64.so.2(+0x104ea)[0x7fae4b44c4ea]
/lib64/ld-linux-x86-64.so.2(+0x105fb)[0x7fae4b44c5fb]
/lib64/ld-linux-x86-64.so.2(+0xcfa)[0x7fae4b43ccfa]
======= Memory map: ========
00400000-0041f000 r-xp 00000000 08:01 5143141                            /home/suijingfeng/source/depth_clustering/build/devel/lib/depth_clustering/show_objects_moosman
0061f000-00620000 r--p 0001f000 08:01 5143141                            /home/suijingfeng/source/depth_clustering/build/devel/lib/depth_clustering/show_objects_moosman
00620000-00621000 rw-p 00020000 08:01 5143141                            /home/suijingfeng/source/depth_clustering/build/devel/lib/depth_clustering/show_objects_moosman
020ff000-02120000 rw-p 00000000 00:00 0                                  [heap]
7fae38000000-7fae38021000 rw-p 00000000 00:00 0 
7fae38021000-7fae3c000000 ---p 00000000 00:00 0 
7fae3c6b6000-7fae3c6bb000 r-xp 00000000 08:01 1056915                    /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7fae3c6bb000-7fae3c8ba000 ---p 00005000 08:01 1056915                    /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7fae3c8ba000-7fae3c8bb000 r--p 00004000 08:01 1056915                    /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7fae3c8bb000-7fae3c8bc000 rw-p 00005000 08:01 1056915                    /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7fae3c8bc000-7fae3c8df000 r-xp 00000000 08:01 1057462                    /usr/lib/x86_64-linux-gnu/libgraphite2.so.3.0.1
7fae3c8df000-7fae3cade000 ---p 00023000 08:01 1057462                    /usr/lib/x86_64-linux-gnu/libgraphite2.so.3.0.1
7fae3cade000-7fae3cae0000 r--p 00022000 08:01 1057462                    /usr/lib/x86_64-linux-gnu/libgraphite2.so.3.0.1
7fae3cae0000-7fae3cae1000 rw-p 00024000 08:01 1057462                    /usr/lib/x86_64-linux-gnu/libgraphite2.so.3.0.1
7fae3cae1000-7fae3caec000 r-xp 00000000 08:01 1057634                    /usr/lib/x86_64-linux-gnu/libjbig.so.0
7fae3caec000-7fae3cceb000 ---p 0000b000 08:01 1057634                    /usr/lib/x86_64-linux-gnu/libjbig.so.0
7fae3cceb000-7fae3ccec000 r--p 0000a000 08:01 1057634                    /usr/lib/x86_64-linux-gnu/libjbig.so.0
7fae3ccec000-7fae3ccef000 rw-p 0000b000 08:01 1057634                    /usr/lib/x86_64-linux-gnu/libjbig.so.0
7fae3ccef000-7fae3cd10000 r-xp 00000000 08:01 4330043                    /lib/x86_64-linux-gnu/liblzma.so.5.0.0
7fae3cd10000-7fae3cf0f000 ---p 00021000 08:01 4330043                    /lib/x86_64-linux-gnu/liblzma.so.5.0.0
7fae3cf0f000-7fae3cf10000 r--p 00020000 08:01 4330043                    /lib/x86_64-linux-gnu/liblzma.so.5.0.0
7fae3cf10000-7fae3cf11000 rw-p 00021000 08:01 4330043                    /lib/x86_64-linux-gnu/liblzma.so.5.0.0
7fae3cf11000-7fae3cf15000 r-xp 00000000 08:01 4330160                    /lib/x86_64-linux-gnu/libuuid.so.1.3.0
7fae3cf15000-7fae3d114000 ---p 00004000 08:01 4330160                    /lib/x86_64-linux-gnu/libuuid.so.1.3.0
7fae3d114000-7fae3d115000 r--p 00003000 08:01 4330160                    /lib/x86_64-linux-gnu/libuuid.so.1.3.0
7fae3d115000-7fae3d116000 rw-p 00004000 08:01 4330160                    /lib/x86_64-linux-gnu/libuuid.so.1.3.0
7fae3d116000-7fae3d118000 r-xp 00000000 08:01 1056904                    /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7fae3d118000-7fae3d318000 ---p 00002000 08:01 1056904                    /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7fae3d318000-7fae3d319000 r--p 00002000 08:01 1056904                    /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7fae3d319000-7fae3d31a000 rw-p 00003000 08:01 1056904                    /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7fae3d31a000-7fae3d37c000 r-xp 00000000 08:01 1056941                    /usr/lib/x86_64-linux-gnu/libXt.so.6.0.0
7fae3d37c000-7fae3d57c000 ---p 00062000 08:01 1056941                    /usr/lib/x86_64-linux-gnu/libXt.so.6.0.0
7fae3d57c000-7fae3d57d000 r--p 00062000 08:01 1056941                    /usr/lib/x86_64-linux-gnu/libXt.so.6.0.0
7fae3d57d000-7fae3d582000 rw-p 00063000 08:01 1056941                    /usr/lib/x86_64-linux-gnu/libXt.so.6.0.0
7fae3d582000-7fae3d583000 rw-p 00000000 00:00 0 
7fae3d583000-7fae3d5a9000 r-xp 00000000 08:01 4330007                    /lib/x86_64-linux-gnu/libexpat.so.1.6.0
7fae3d5a9000-7fae3d7a9000 ---p 00026000 08:01 4330007                    /lib/x86_64-linux-gnu/libexpat.so.1.6.0
7fae3d7a9000-7fae3d7ab000 r--p 00026000 08:01 4330007                    /lib/x86_64-linux-gnu/libexpat.so.1.6.0
7fae3d7ab000-7fae3d7ac000 rw-p 00028000 08:01 4330007                    /lib/x86_64-linux-gnu/libexpat.so.1.6.0
7fae3d7ac000-7fae3f062000 r-xp 00000000 08:01 1057582                    /usr/lib/x86_64-linux-gnu/libicudata.so.55.1
7fae3f062000-7fae3f261000 ---p 018b6000 08:01 1057582                    /usr/lib/x86_64-linux-gnu/libicudata.so.55.1
7fae3f261000-7fae3f262000 r--p 018b5000 08:01 1057582                    /usr/lib/x86_64-linux-gnu/libicudata.so.55.1
7fae3f262000-7fae3f263000 rw-p 018b6000 08:01 1057582                    /usr/lib/x86_64-linux-gnu/libicudata.so.55.1
7fae3f263000-7fae3f284000 r-xp 00000000 08:01 1058282                    /usr/lib/x86_64-linux-gnu/libxcb.so.1.1.0
7fae3f284000-7fae3f483000 ---p 00021000 08:01 1058282                    /usr/lib/x86_64-linux-gnu/libxcb.so.1.1.0
7fae3f483000-7fae3f484000 r--p 00020000 08:01 1058282                    /usr/lib/x86_64-linux-gnu/libxcb.so.1.1.0
7fae3f484000-7fae3f485000 rw-p 00021000 08:01 1058282                    /usr/lib/x86_64-linux-gnu/libxcb.so.1.1.0
7fae3f485000-7fae3f4f3000 r-xp 00000000 08:01 4330105                    /lib/x86_64-linux-gnu/libpcre.so.3.13.2
7fae3f4f3000-7fae3f6f3000 ---p 0006e000 08:01 4330105                    /lib/x86_64-linux-gnu/libpcre.so.3.13.2
7fae3f6f3000-7fae3f6f4000 r--p 0006e000 08:01 4330105                    /lib/x86_64-linux-gnu/libpcre.so.3.13.2
7fae3f6f4000-7fae3f6f5000 rw-p 0006f000 08:01 4330105                    /lib/x86_64-linux-gnu/libpcre.so.3.13.2
7fae3f6f5000-7fae3f6fc000 r-xp 00000000 08:01 1057287                    /usr/lib/x86_64-linux-gnu/libffi.so.6.0.4
7fae3f6fc000-7fae3f8fb000 ---p 00007000 08:01 1057287                    /usr/lib/x86_64-linux-gnu/libffi.so.6.0.4
7fae3f8fb000-7fae3f8fc000 r--p 00006000 08:01 1057287                    /usr/lib/x86_64-linux-gnu/libffi.so.6.0.4
7fae3f8fc000-7fae3f8fd000 rw-p 00007000 08:01 1057287                    /usr/lib/x86_64-linux-gnu/libffi.so.6.0.4
7fae3f8fd000-7fae3f961000 r-xp 00000000 08:01 1057853                    /usr/lib/x86_64-linux-gnu/libpcre16.so.3.13.2
7fae3f961000-7fae3fb61000 ---p 00064000 08:01 1057853                    /usr/lib/x86_64-linux-gnu/libpcre16.so.3.13.2
[... 145 lines cut ...]
7fae4645b000-7fae4645f000 r--p 00103000 08:01 1061344                    /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.58.0
7fae4645f000-7fae46460000 rw-p 00107000 08:01 1061344                    /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.58.0
7fae46460000-7fae46477000 r-xp 00000000 08:01 1057052                    /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.58.0
7fae46477000-7fae46676000 ---p 00017000 08:01 1057052                    /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.58.0
7fae46676000-7fae46677000 r--p 00016000 08:01 1057052                    /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.58.0
7fae46677000-7fae46678000 rw-p 00017000 08:01 1057052                    /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.58.0
7fae46678000-7fae47d77000 r-xp 00000000 08:01 5390698                    /opt/ros/kinetic/lib/libopencv_imgproc3.so.3.1.0
7fae47d77000-7fae47f76000 ---p 016ff000 08:01 5390698                    /opt/ros/kinetic/lib/libopencv_imgproc3.so.3.1.0
7fae47f76000-7fae47f7f000 r--p 016fe000 08:01 5390698                    /opt/ros/kinetic/lib/libopencv_imgproc3.so.3.1.0
[... cut ...]
7fae48788000-7fae48988000 ---p 00172000 08:01 1050753                    /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
7fae48988000-7fae48992000 r--p 00172000 08:01 1050753                    /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
7fae48992000-7fae48994000 rw-p 0017c000 08:01 1050753                    /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
7fae48994000-7fae48998000 rw-p 00000000 00:00 0 
7fae48998000-7fae48ff1000 r-xp 00000000 08:01 1053701                    /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.5.1Aborted (core dumped)

niosus, 2016-12-20:
This is related to Qt5 vs Qt4. This code uses QGlViewer, which is built with Qt4 if you install it from apt-get on Ubuntu. If you try to build the code with Qt5 it links to wrong libraries. Please make sure you have Qt4 installed and build again.

suijingfeng, 2016-12-20:
Thank you reply very much， I‘m a Chinese student， My Ubuntu 16.04 have both qt4 and qt5 installed. and when I compile you source code , the program link the qt5 automatically, but I have no idea how to specify  the target link qt4 properly.

niosus, 2016-12-20:
I haven't tested it on Ubuntu 16.04 as we still have 14.04 in the lab. It should, however, run on 16.04. I will check it an will come back to this question in the coming days.

niosus, 2016-12-22:
I have tried this on by ubuntu 16.04 and confirm the issue. I spent an hour to fix this, but something keeps linking some qt5 libs to the binary and I haven't tracked that down. I will fix this either later today or on Monday.

niosus, 2017-01-08:
I am sorry for such a delay. Had too much on my plate during the holidays. Anyway, I have fixed this. We will now link to Qt4 on Ubuntu 14.04 and to Qt5 on Ubuntu 16.04. I have tested this on my Ubuntu 16.04. 

There seems to be an unrelated bug with the representation of the objects, but at least it is running. I will investigate further as soon as I reach the lab.

Again, thanks for reporting.

commit 8a764ab1 (2017-01-08): Fix #1

We now link to qt4 on ubuntu 14.04 and to qt5 on ubuntu 16.04
closed 2017-01-08 by niosus