PRBonn/depth_clustering #5: Segmentation fault (core dumped) occurred when running ROS node

TixiaoShan, 2017-03-10:
Hi,
Thanks for sharing the source code of your paper. I downloaded and compiled successfully. The real world data "Frank Moosmann's "Velodyne SLAM" Dataset" works like a charm

However, when I run the ros node "show_objects_node" and playing my own bag file that is publishing velodyne point cloud, the node terminates with such error:

```
./show_objects_node --num_beams 16

 ===================== Setting Connection =====================
|| Sender: CloudOdomRosSubscriber (id: 1)
|| Type:   STREAMER
|| 				|
|| 				V
|| Client: DepthGroundRemover (id: 4)
 ==============================================================

 ===================== Setting Connection =====================
|| Sender: DepthGroundRemover (id: 4)
|| Type:   STREAMER
|| 				|
|| 				V
|| Client: ImageBasedClustere<(short)1, (short)1> > (id: 6)
 ==============================================================

 ===================== Setting Connection =====================
|| Sender: ImageBasedClustere<(short)1, (short)1> > (id: 6)
|| Type:   STREAMER
|| 				|
|| 				V
|| Client: ObjectPtrStorer (id: 3)
 ==============================================================

 ===================== Setting Connection =====================
|| Sender: CloudOdomRosSubscriber (id: 1)
|| Type:   STREAMER
|| 				|
|| 				V
|| Client: Visualizer (id: 2)
 ==============================================================
INFO: Running with angle tollerance: 10.000000 degrees
INFO: Ground removed in 815 us
ERROR: DiffType is NONE. Please set it.
Segmentation fault (core dumped)
```
I don't see any settings for DiffType in the .cpp file. Any idea what's going on here?

Thanks!

TixiaoShan, 2017-03-10:
I added 
`clusterer.SetDiffType(DiffFactory::DiffType::ANGLES);`
after declare cluster object.

niosus, 2017-03-11:
Yeah, sorry for that. I have been changing the interface and did not change the examples. I will reopen the issue until I do it. Thanks for reporting!

closed 2017-03-14 by niosus