PRBonn/depth_clustering #6: ｑｕｅｓｔｉｏｎｓ　in the ground remove.cpp

ttomchy, 2017-04-18:
hello ,thanks for you to make your code available .I have forked your code ,however in the depth_ground_remover.cpp ,there is a function named CreateAngleImage(),why do you turn a depth image into the image with angle ? and then use  an  ApplySavitskyGolaySmoothing() ｆｕｎｃｔｉｏｎ　to smooth the angle image .I am confusing about it  so I wonder that may you give  more information about it ?

niosus, 2017-04-18:
The angles are the measure I use to remove ground. I use smoothing because otherwise the readings are too noisy because of the sensor noise. Please read the paper for more details.

closed 2017-04-18 by niosus

ttomchy, 2017-04-24:
OK ,Thank you.  in the code you also provide the examples of the kitti ,I wonder that if you can share the kitti dataset or share the link of the kitti dataset that you use.