PRBonn/depth_clustering #10: Errors during compiling

RabahZeroukhi, 2017-06-26:
Hello, 
I tried to compile the package using the tool catkin_make in workspace path and I had these errors

[image]

any help will be appreciated

niosus, 2017-06-26:
Please install `catkin_tools`: 
```
supo pip install catkin_tools
```
and run `catkin build` instead of `catkin_make`. It will probably produce either a proper build or a better error. Also, make sure that you have `eigen3` library installed.

RabahZeroukhi, 2017-06-28:
thanks for reply, I want to build using cmake .. and make -j4, it was configured well after editing Cmakelist.txt removing "find_package(eigen3 required)" and adding
 "find_package(PkgConfig)
pkg_search_module(Eigen3 REQUIRED eigen3)",
because i had the problem which you mentioned (eigen 3 lib)
[image]
and then when i run " make -j4" it didn't compile the whole package 
[image]

for you information, i have ubuntu 14.04 as OS and  ROS indigo. thanks again.

niosus, 2017-06-28:
It looks like smth is wrong with your system. `eigen3` should be found on a system with ROS indigo out of the box using `find_package(Eigen3 REQUIRED)` (make sure to spell `Eigen3` *exactly* like here). 

To check this, I have just spawned a new docker image with image: `ros:indigo-perception` and after following the installation steps everything builds as expected using `cmake` and `make`. You can also do it yourself if you have docker running on your system. Here are all commands:

Run the docker container:
```
sudo docker run -it ros:indigo-perception bash
```
Now from this container install the libraries missing out of the box:
```
apt update
apt install -y libqglviewer-dev freeglut3-dev
```
Build the actual code:
```
mkdir -p catkin_ws/src && cd catkin_ws/src
git clone https://github.com/Photogrammetry-Robotics-Bonn/depth_clustering.git
cd depth_clustering && mkdir build && cd build
cmake ..
make -j4
```
This builds the code for me with no issues and proves that the code builds on a clean system. Therefore, I feel that there is smth strange going on on your side. I can't really tell you more, I guess...

RabahZeroukhi, 2017-07-06:
Yes it works, thanks very much.

closed 2017-07-06 by RabahZeroukhi

LiShuaixin, 2018-11-26:
HI @niosus 
I still meet the error while I use catkin build instead of catkin_make.

any help will be appreciated!

LiShuaixin, 2018-11-26:
> HI @niosus
> I still meet the error while I use catkin build instead of catkin_make.
> [image]
> 
> any help will be appreciated!

It can be solved by adding 
`FIND_PACKAGE(GTest)`
and repleacing 
`ADD_SUBDIRECTORY(/usr/src/gtest/ gtest)`

anyone could refer to [Stack Overflow](https://stackoverflow.com/questions/42085151/cmake-conflict-with-multiple-gtest) to find the principle behind of this case.

tornadomeet, 2019-03-28:
@LiShuaixin 
still the same error after doing
```shell
It can be solved by adding
FIND_PACKAGE(GTest)
and commenting
ADD_SUBDIRECTORY(/usr/src/gtest/ gtest)
```