PRBonn/depth_clustering #13: Writing points depth to depth_image if it is further than current depth.

zzzdeb, 2017-09-19:
Hi,
First: Thank you a lot for providing your code.
Second: isnt it supposed to be otherwise ">"? in

https://github.com/Photogrammetry-Robotics-Bonn/depth_clustering/blob/master/src/projections/spherical_projection.cpp

auto& current_written_depth = this->_depth_image.template at<float>(bin_rows, bin_cols);
if (current_written_depth < dist_to_sensor) {
      // write this point to the image only if it is closer
current_written_depth = dist_to_sensor;
}

greetings,
Zolboo.

niosus, 2017-09-20:
I believe you are right. The only thing is that some care might be taken because I initialize all values with 0. Thanks. I will look into it, but probably in a week or so.

rambleramble, 2017-12-21:
can we confirm if it should be ">"? Thanks!

niosus, 2017-12-23:
I am sorry for a huge delay. I have looked a bit into this and eventually, you can think of it as a convention. We either pick the furthest point that falls to the same pixel of the depth image or the closest one. The way it is coded in this method makes it easy to check for depth validity as all invalid depth readings can be set to 0, thus avoiding using some arbitrary `kDepthInvalid` value.

niosus, 2017-12-23:
I guess I will close this issue for now.

closed 2017-12-23 by niosus