PRBonn/depth_clustering #14: segmentation fault when running show_objects_node

bgheneti, 2017-10-17:
The node runs for a couple of seconds showing clusters then crashes with a segmentation fault. The amount of time take to crash varies a little 0.5-8 seconds, but never very long.

I am using a VLP16 lidar and am playing back a bag file with PointCloud2 data.

Here is the console I get before the crash 

```
INFO: Running with angle tollerance: 10.000000 degrees 
[INFO]: Not correcting depth data.
INFO: Ground removed in 10392 us
INFO: image based labeling took: 6617 us
INFO: labels image sent to clients in: 18 us
INFO: prepared clusters in: 2624 us
INFO: clusters shared: 222 us
[INFO]: Not correcting depth data.
INFO: Ground removed in 911 us
INFO: image based labeling took: 2328 us
INFO: labels image sent to clients in: 8 us
INFO: prepared clusters in: 943 us
INFO: clusters shared: 83 us
[INFO]: Not correcting depth data.
INFO: Ground removed in 5766 us
INFO: image based labeling took: 12329 us
INFO: labels image sent to clients in: 27 us
INFO: prepared clusters in: 4558 us
INFO: clusters shared: 451 us
[INFO]: Not correcting depth data.
INFO: Ground removed in 1603 us
INFO: image based labeling took: 4190 us
INFO: labels image sent to clients in: 26 us
INFO: prepared clusters in: 2142 us
INFO: clusters shared: 401 us
[INFO]: Not correcting depth data.
INFO: Ground removed in 887 us
INFO: image based labeling took: 2235 us
INFO: labels image sent to clients in: 8 us
INFO: prepared clusters in: 858 us
INFO: clusters shared: 103 us
[INFO]: Not correcting depth data.
INFO: Ground removed in 5480 us
INFO: image based labeling took: 9984 us
INFO: labels image sent to clients in: 30 us
INFO: prepared clusters in: 3072 us
INFO: clusters shared: 186 us
Segmentation fault (core dumped)
```

niosus, 2017-10-25:
Wow, this is strange. Sorry that it crashes on you. I cannot tell you why it crashes, but you should be able to compile it in debug mode and then run within gdb or lldb to get the full stack trace. If I have that I will be able to help.

niosus, 2018-02-06:
Closing for lack of activity.

closed 2018-02-06 by niosus

saimanoj18, 2018-04-10:
I too have ended up with Segmentation fault and here is the gdb back trace

`[INFO]: Not correcting depth data.
INFO: Ground removed in 992 us
INFO: image based labeling took: 2158 us
INFO: labels image sent to clients in: 8 us
INFO: prepared clusters in: 1106 us
INFO: clusters shared: 104 us
[INFO]: Not correcting depth data.
INFO: Ground removed in 1247 us
INFO: image based labeling took: 2446 us
INFO: labels image sent to clients in: 4 us
INFO: prepared clusters in: 1290 us
INFO: clusters shared: 104 us
[INFO]: Not correcting depth data.
INFO: Ground removed in 1012 us
INFO: image based labeling took: 2548 us
INFO: labels image sent to clients in: 68 us
INFO: prepared clusters in: 1970 us
INFO: clusters shared: 99 us

Thread 12 "show_objects_no" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffb63a7700 (LWP 12290)]
0x00007ffff7ba748c in depth_clustering::CloudOdomRosSubscriber::RosCloudToCloud(boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&)
    () from /home/sai/depth_clustering-master/build/devel/lib/libros_bridge.so
(gdb) bt
#0  0x00007ffff7ba748c in depth_clustering::CloudOdomRosSubscriber::RosCloudToCloud(boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&) () from /home/sai/depth_clustering-master/build/devel/lib/libros_bridge.so
#1  0x00007ffff7ba76cb in depth_clustering::CloudOdomRosSubscriber::CallbackVelodyne(boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&) () from /home/sai/depth_clustering-master/build/devel/lib/libros_bridge.so
#2  0x00007ffff7bb615e in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const>) () from /home/sai/depth_clustering-master/build/devel/lib/libros_bridge.so
#3  0x00007ffff7bc8739 in message_filters::CallbackHelper1T<boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&, sensor_msgs::PointCloud2_<std::allocator<void> > >::call(ros::MessageEvent<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&, bool) ()
   from /home/sai/depth_clustering-master/build/devel/lib/libros_bridge.so
#4  0x0000000000426e27 in message_filters::Subscriber<sensor_msgs::PointCloud2_<std::allocator<void> > >::cb(ros::MessageEvent<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&) ()
#5  0x000000000042ec86 in ros::SubscriptionCallbackHelperT<ros::MessageEvent<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&, void>::call(ros::SubscriptionCallbackHelperCallParams&) ()
#6  0x00007ffff6ee6e2d in ros::SubscriptionQueue::call() () from /opt/ros/kinetic/lib/libroscpp.so
#7  0x00007ffff6e8c6f8 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) () from /opt/ros/kinetic/lib/libroscpp.so
#8  0x00007ffff6e8e0fb in ros::CallbackQueue::callAvailable(ros::WallDuration) () from /opt/ros/kinetic/lib/libroscpp.so
#9  0x00007ffff6eea344 in ros::AsyncSpinnerImpl::threadFunc() () from /opt/ros/kinetic/lib/libroscpp.so
#10 0x00007ffff01b75d5 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
#11 0x00007ffff734a6ba in start_thread (arg=0x7fffb63a7700) at pthread_create.c:333
#12 0x00007ffff452441d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
`

saimanoj18, 2018-04-10:
Following this recommendation has solved the issue

https://github.com/Photogrammetry-Robotics-Bonn/depth_clustering/pull/18

The reason can be that I am using Robosense Lidar and not Velodyne lidar and hence their field definitions can be different!

LiShuaixin, 2018-11-26:
Hi, there. I try to show objects in moosmann dataset but failed. 

And I got some feedback as follow:
`
INFO: Reading from: /home/lee/depth_clustering_ws/data/scenario1/ 
INFO: running on Moosman data
INFO: Getting file paths from folder: /home/lee/depth_clustering_ws/data/scenario1/
INFO: There are 2513 '.png' files in the folder.
INFO: Getting file paths from folder: /home/lee/depth_clustering_ws/data/scenario1/
INFO: There are 1 'img.cfg' files in the folder.
INFO: Set en_US.UTF-8 locale.
INFO: Reading config.
INFO: Skipping commentary: 
	 # imgHSize; imgVSize; horizStartAngle(grad); horizStopAngle(grad); vertAngle_1; vertAngle_2; ...; vertAngle_n
start:180.000000, stop:-180.000000, span:360.000000, step:-0.413793
INFO: Params sucessfully read. Rows: 64, Cols: 870

 ===================== Setting Connection =====================
|| Sender: DepthGroundRemover (id: 3)
|| Type:   STREAMER
|| 				|
|| 				V
|| Client: ImageBasedClustere<(short)1, (short)1> > (id: 5)
 ==============================================================

 ===================== Setting Connection =====================
|| Sender: ImageBasedClustere<(short)1, (short)1> > (id: 5)
|| Type:   STREAMER
|| 				|
|| 				V
|| Client: ObjectPtrStorer (id: 2)
 ==============================================================
show_objects_moosmann: /usr/local/include/eigen3/Eigen/src/Core/DenseStorage.h:128：Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = float; int Size = 16; int MatrixOrArrayOptions = 0]: Assertion ‘(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"’ Failed。`

I checked the web and found it's the problem caused by Eigen. Is there anyone know how to fix it? Many thanks!