PRBonn/depth_clustering #16: pointcloud convert to range image, is it always 64x870?

rambleramble, 2017-12-21:
Hi, thank you first for sharing this excellent work!

If my input is pointcloud, such as the  KITTI .BIN file,  is this conversion using projection (sphere or ring),  always conver to 64x870 size?

How/Where this 870 is calculated? 
And can we change this size 870, I saw some other size such as 64x1030.

My concern is the original xyz of the poincloud from BIN is around ~120,000 points while the range image is only covering  55680 xyz points (55680=870*64).  It seems more than 50% points are removed/overwrite in this "pc2range" conversion process. Can you comment on this?

Thank you very much!

niosus, 2017-12-23:
So, the way it is implemented now was tailored to Frank Moosman's data. These data are all 64x870, which seems to give a good tradeoff between speed and precision. The way I handle multiple points that fall into the same pixel is that only the *furthest* reading is taken per pixel, but all the points that fall into it get assigned the same label. It can create some minor artifacts, but seemed to work well in practice for me. 

You can however, pick any resolution you want if you look into the `ProjectionParams` class by adding another function like [this](https://github.com/Photogrammetry-Robotics-Bonn/depth_clustering/blob/master/src/projections/projection_params.cpp#L187) one with any resolution you want.

Be advised, that working with KITTI data directly might be a little imprecise as the HDL64 has two sets of beams that do not exactly match onto a single depth image.

closed 2017-12-23 by niosus

tornadomeet, 2019-04-29:
i think KITTI should be set as:
```c++
params.SetSpan(SpanParams(-180_deg, 180_deg, 4000),
SpanParams::Direction::HORIZONTAL);
```
because Velodyne HDL-64E has 0.09 degree angular resolution.

lonlonago, 2019-06-17:
@niosus 
1.  I also think HDL64 row step should be 4000 or 2048,  but will this increase runtime a lot ?
2.  can you  explain further about "HDL64 has two sets of beams that do not exactly match onto a single depth image" 
3. the AngleDiff::PreComputeAlphaVecs,  if the angle is add by step, then it's a constant value, why need a vector to store and obtain it by r,c indices ?

thanks for your work and reply!