PRBonn/depth_clustering #17: how can this publish topics represent the bounding box?

weisongwen, 2018-02-07:
Hi  
Thanks for your work on this package,
how can the bounding box be published as a topic after the segmentation?
after ``catkin build``, it seems that they is no rosnode related to  the following two file:
``
1. save_clusters_node.cpp
2.show_objects_node.cpp
``
Best 
Welson

niosus, 2018-02-07:
Yes, this will need to be implemented as I did not focus on publishing anything through ros. I only use ros as a method to deliver me data.

closed 2018-02-07 by niosus

weisongwen, 2018-02-07:
Hi, @niosus 
Do you have any ideas on this issue?  how can we get the bounding box information and then publish them?
Best 
Welson

niosus, 2018-02-07:
You can look where the visualizer gets the clusters (e.g. like [here](https://github.com/Photogrammetry-Robotics-Bonn/depth_clustering/blob/master/examples/ros_nodes/show_objects_node.cpp#L102) and write a similar component that would receive the clusters and create bounding boxes and send them through ros.

weisongwen, 2018-02-07:
Hi, @niosus 
I tried this and got the value, but once the main function run to the following code:
``auto exit_code = application.exec();``
the code blocked in  **application.exec()** and I can not jump out of application.exec() to publish the topic.
Do you have any advice on this ?

Best 
Welson

niosus, 2018-02-07:
Please make sure you understand everything that is going on in the file, especially [this line](https://github.com/Photogrammetry-Robotics-Bonn/depth_clustering/blob/master/examples/ros_nodes/show_objects_node.cpp#L109)

You have to be comfortable with understanding where and why a thread is spawned. I will not be able to explain it here fully, but the main idea is that there is a ros thread and a qt thread. The main thread of the application is owned by qt, while ros gets a new thread to own. You must implement a component that connects to other by using `AddClient(...)` functions. This way, the whole pipeline will be triggered by ros receiving data. This data will follow the path through all the components and must end up with your new components that will publish the data through ros again.

weisongwen, 2018-02-07:
Thanks for your kind advice, it is clear to me, I will implement this!

Best 
Welson

didriksg, 2018-03-13:
Hi!
Did you ever implement this @weisongwen  ? I would be very interested in using the published bounding boxes myself!

weisongwen, 2018-03-14:
Hi, @didriksg  
I have not implement this yet, as i am busy with some other things.

I may conduct this next month,
Best 
Welson

didriksg, 2018-03-14:
Hi @weisongwen 
That's alright! If I get the time, I might just look into implementing it myself.
Thank you for your quick answer! :)

PanZhichen, 2018-04-03:
@weisongwen Hello, I have the same confusion with you. Have you found a good way to publish the bounding boxes or clusters as a ROS topic? Thanks for your help :)

weisongwen, 2018-04-03:
hi, @PanZhichen 
This function have not been implement yet,
Best 
Welson

weisongwen, 2018-04-04:
Hi @PanZhichen  you can refer to the msg format provided in Autoware,
Best 
Welson

AlizadehAli, 2019-07-26:
Hi, @PanZhichen and @weisongwen , Did you publish this bounding boxes ? I am trying to publish but i cannot find the way. Thanks for help !
Best 
Elif

ElifAksu, 2019-07-26:
Hi @PanZhichen  and @weisongwen ,
Did you publish the bounding box? I cannot find the way. Could you please share your method? Thanks for help!
Best,
Elif