PRBonn/depth_clustering #19: subscribing to clusters/bounding boxes

bgheneti, 2018-03-11:
Thanks for creating this awesome tool!

I noticed that it can subscribe to ros topics to obtain pointclouds to cluster, but was wondering if there are any ways to subscribe to the bounding boxes or clusters as well.

If not, how would you recommend going about doing this?

Thanks,
Banti

niosus, 2018-03-12:
Hello @bgheneti and thanks! 
I don't fully understand what you mean by "subscribing to the bounding boxes and clusters as well". You mean that somebody would send them and this code would subscribe? Or do you mean that some other code would subscribe to the ones produces by this code?

I don't really see the reason for the first use case, but the second should be relatively easy to implement by introducing some publisher class that would receive the clusters and send their bounding boxes to somebody else.

bgheneti, 2018-03-12:
I mean the second use case. Thanks! Will look into it.

closed 2018-03-12 by bgheneti

PanZhichen, 2018-04-03:
@bgheneti Hello, I have the same confusion with you. Have you found a good way to publish the bounding boxes or clusters as a ROS topic? Thanks for your help :)

bgheneti, 2018-04-03:
Hi @PanZhicken. I added a publisher under ros_bridge that sends out a pointcloud2 with associated cluster labels on the points. This is on a local computer and I can share that as a fork in the next day or two if that would be helpful.

PanZhichen, 2018-04-04:
Dear @bgheneti, thank you so much for helping me work this out. And please send me the link when you release your source could, thanks again.

bgheneti, 2018-04-06:
@PanZhichen here is my updated fork: https://github.com/bgheneti/depth_clustering. Lmk if you have any questions. @niosus I can also merge some portion of the changes into the main branch if that would be helpful.

niosus, 2018-04-07:
@bgheneti sure, just throw in some PRs and we'll see how it goes :)

hecperleo, 2018-04-23:
As everyone is saying, thanks a lot for your awesome work!
@PanZhichen Can you explain how to use your modifications? I run first a .bag with /velodyne_points and `rosrun depth_clustering show_objects_node --n_beams 32` is working fine without your modifications. If I use your fork I get this output:

```
 rosrun depth_clustering show_objects_node --num_beams 32

 ===================== Setting Connection =====================
|| Sender: CloudOdomRosSubscriber (id: 1)
|| Type:   STREAMER
|| 				|
|| 				V
|| Client: DepthGroundRemover (id: 4)
 ==============================================================

 ===================== Setting Connection =====================
|| Sender: DepthGroundRemover (id: 4)
|| Type:   STREAMER
|| 				|
|| 				V
|| Client: ImageBasedClustere<(short)1, (short)1> > (id: 6)
 ==============================================================

 ===================== Setting Connection =====================
|| Sender: ImageBasedClustere<(short)1, (short)1> > (id: 6)
|| Type:   STREAMER
|| 				|
|| 				V
|| Client: ObjectPtrStorer (id: 3)
 ==============================================================

 ===================== Setting Connection =====================
|| Sender: CloudOdomRosSubscriber (id: 1)
|| Type:   STREAMER
|| 				|
|| 				V
|| Client: Visualizer (id: 2)
 ==============================================================
INFO: Running with angle tollerance: 5,000000 degrees
[ERROR] [1524486854.347886270]: FUDGE

```

bgheneti, 2018-04-23:
@hecperleo fixed

hecperleo, 2018-05-03:
@bgheneti @niosus Sorry for asking here, but I have a small question related. It is possible to point to a cluster? I want to calculate distance to one or more clusters, but I dont know how to point to them or if it is possible. Thanks.

PanZhichen, 2018-05-16:
Hello @bgheneti , thanks for your help:-), and other questions need your help:
- I used **'git clone'** to download your modified code and changed **'sub_topic_clouds'** to **'/velodyne_points'** in **'publish_clusters_node.cpp'**.  Then used **‘catkin build depth_cluster’** to compile the project.
- Then I **rosrun depth_clustering publish_clusters_node**, and start replaying a bag file which contains **/velodyne_points** topic.

**BUT** the visualization window didn't appear, and nothing was output when I **'rostopic echo /depth_clusterer/segmented_cloud'**

@bgheneti @hecperleo Could you me give me some suggestions about what should I do? Thanks very much!

bgheneti, 2018-07-13:
Hi @hecperleo you can calculate cluster centroid from the PointCloud2 message using PCL. The distance to this in the boat reference frame should be what you want.

firesnow1234, 2019-01-08:
> Thanks you! I didn't realize that you changed the subscriber topic_clouds. Now is working fine.

> @PanZhichen here is my updated fork: https://github.com/bgheneti/depth_clustering. Lmk if you have any questions. @niosus I can also merge some portion of the changes into the main branch if that would be helpful.

Hi, thanks  for your share.  I run a ros node which send a pointcould2 topic named "/points_raw_map_crop", but there's no response in node "publish_clusters_node", it seems still waiting for  the right topic.  Can you tell me how to publish the right topic?

urbansound8K, 2022-01-11:
@firesnow1234 I have the same here! could you able to solve it?
@hecperleo @bgheneti  ? could you please help to check, PLEASE?