PRBonn/depth_clustering #20: Tuning

didriksg, 2018-03-13:
Hello!
Thank you for creating this awesome package! I wondered if there is a possibility of tuning the clustering parametres, and basicly influence what that qualifies as a bounding box and not, as I am mostly interested in small objects, and want to remove large objects, or objects that does not qualify to my terms.

Is this already a possibility? If not, could you point me in the right direction?

Thanks!

niosus, 2018-03-14:
Hey @didriksg 
So there are two things we can talk about. First one is the separation criterium for clustering (angle beta in the paper and code). This decides how likely it is to separate clusters. Now, this should be picked once for your sensor if you care about meaningful clusters.

The other thing is what is considered to be a cluster, as you put it. To visualize the clusters and make it look decent, I decided to not show very small clusters. This was not indended to be passed to any other human and was done purely for visualization, therefore it is a hack.
In the viewer I just don't draw anything that has height of less than `0.3`: https://github.com/Photogrammetry-Robotics-Bonn/depth_clustering/blob/2107024458f639af5a33c25d8bb3cc2f7a9d93bc/src/qt/drawables/drawable_cube.cpp#L9

I also draw big objects with different color for the same reason: 
https://github.com/Photogrammetry-Robotics-Bonn/depth_clustering/blob/2107024458f639af5a33c25d8bb3cc2f7a9d93bc/src/qt/drawables/drawable_cube.cpp#L13-L18

So to adjust that you will have to modify the code as there is no easy way for that. I expected the users to receive the clusters and then post-process them to their needs.

closed 2018-03-14 by niosus

didriksg, 2018-03-14:
Ah, I see! Thank you for your quick answer and good explenation. I will try it out later this week.