PRBonn/depth_clustering #21: clustering problems

FlexiPro, 2018-03-23:
Hello,

First of all, very nice tool! 
I have been testing these nodes with our VLP16 and they worked very well. We drove around with our car and had similar results as in your movies. Now we have created a simulator that calculates the interference of the lidar rays and some objects. This simulator creates pointclouds as the VLP16 would do, but I don't get the nice results as on the tests with the real VLP16. I'm not sure what is going wrong, but maybe you have an idea.

The simulated environment is a flat surface with some vertical walls and every wall should in fact be seen as one cluster. I don't see any clustering going on if I do not set the minimal cluster size to 10.  But then I see that the calculated clusters are only like vertical small boxes. Is there a limitation towards how far the points can be apart form each other and is this direction dependent? I have put some screenshots below.

I appreciate your help very much!

[image]

[image]

[image]

[image]

niosus, 2018-03-23:
Thanks for the kind words, means a lot to me :)
As for your issue. I would probably start by visualizing range images and making sure they do not contain gaps. My guess is that the range images become sparse in lateral direction and this causes the artifacts you are seeing.

niosus, 2018-03-23:
I will close this issue, but let's continue discussing it here if you wish.

closed 2018-03-23 by niosus

FlexiPro, 2018-03-26:
Hello Igor,

Sorry for my late reply. Thanks that you want to discuss this a little bit further.

I don’t see extremely large gaps between the points, I have visualized the pointcloud in rviz. (see below) But is there a limitation on how far the points can be apart from each other? I was looking for a setting like that in your code, but did not immediately find such setting. I also tried differentiating on ‘line distance’ instead of on ‘angle’, but that did not give much better results. What parameters would you advise me to play with in order to get this clustering going? And what steps woud you do in order to analyse this problem?

Thank you in advance!
Best regards,
Glenn Aesaert

[cid:image001.png@01D3C50E.2FD02320]

FlexiPro, 2018-03-28:
Hello Igor,

Maybe another update from my side. I got the chance to work a little bit further on this today.

I got the clustering working by changing the STEP_ROW and STEP_COL to 10 (it was 1 before) in the ‘linear_image_labeler.h’-file. If I’m correct this gives the labeler a wider area to detect clusters in. I think the biggest issue is that our objects are always diagonal positioned in the image, as the clustering is more based on orthogonal objects. Is this correct? The problem that I see is that the bounding boxes are not drawn optimal with this solution. And that the clusters sometimes are not made completely because they are prohibited by another ‘smaller’ cluster. (you can see this in the image below)

It would be nice if the clustering is independent from the orientation of the objects. Is this something that is available in your code also and that needs to be activated, or does this need to be implemented also? Do you think this is difficult to make or not? (I don’t have that much experience with clustering algorithms)

Best regards,
Glenn Aesaert

[cid:image002.png@01D3C6AA.734F8E90]

[cid:image003.png@01D3C6AA.734F8E90]

niosus, 2018-03-28:
@FlexiPro I feel like you are trying to solve something with this method that this method never intended to solve. As far as I understand you are trying to cluster arbitrary objects positioned arbitrarily in the world. The method does NOT target this. It was written to process data coming from Velodyne. It is assumed that one knows the positioning of the lasers and therefore *exactly* how big are the gaps *in the range image* between the points.