PRBonn/depth_clustering #26: "segment fault" and "[INFO]: Not correcting depth data"

JSRGFJZ, 2018-08-11:
Hi, I really thank you for your great work, now, when I compile itand run it, I alwayes get  "[INFO]: Not correcting depth data". this is problem 1.
for problem 2: the original compile way is using cmake, however I alter it to bazel, and I succeed, but when I run "./show_objects_node --num_beams 16", I alwayes get "[1]    30971 segmentation fault (core dumped)  ./show_objects_node --num_beams 16", sorry to bother you.
hope to get your help.

niosus, 2018-08-11:
1. This is not an error, just an info message.
2. Make sure you have the data that is structured as expected. See other issues in the repo regarding segfaults. I expect a certain structure of pointcloud2 messages and the code does not check if this is not the case.

closed 2018-08-11 by niosus

JSRGFJZ, 2018-08-13:
@niosus Hi, may I ask again that how do I set _corrections in FixDepthSystematicErrorIfNeeded, I notice that you add the interface but you didn't use it.

JSRGFJZ, 2018-08-13:
OK, thanks for that, I also notice that you write in your thesis you manually create 30 different label-out 3D data. Would you like to share them, I'd like to analize the performence of under-segmentation and over-segmentation.
Really thank you for your help.

j.z.feng@foxmail.com
 
From: Igor Bogoslavskyi
Date: 2018-08-11 18:52
To: PRBonn/depth_clustering
CC: Jean; Author
Subject: Re: [PRBonn/depth_clustering] "segment fault" and "[INFO]: Not correcting depth data" (#26)
This is not an error, just an info message.
Make sure you have the data that is structured as expected. See other issues in the repo regarding segfaults. I expect a certain structure of pointcloud2 messages and the code does not check if this is not the case.