PRBonn/depth_clustering #30: Segfault happens while running show_object_node with KITTI rosbag. show_object_kitti works fine with .bin files of the same sequence.

tuandle, 2018-11-17:
Hi @niosus , 
When I tried to run `show_object_node` with a KITTI rosbag, I have a segfault. However, if I use `show_object_kitti` with the raw data of the same sequence, it works fine. 
Here is the [link](https://youtu.be/30N6brOfOMc) to the screen video. 
I downloaded the kitti rosbag 2011_09_30_drive_18.bag from [here](http://robotics.ethz.ch/~asl-datasets/segmap/segmap_data/kitti/) and same sequence from the KITTI official website.
The full trace back by gdb is:
```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffc097f700 (LWP 30258)]
0x00007ffff7b485a4 in depth_clustering::BytesTo<unsigned short> (data=..., start_idx=2835384544)
    at /home/tuan/workshop_ws/src/depth_clustering/src/ros_bridge/cloud_odom_ros_subscriber.cpp:45
45	    byte_array[i] = data[start_idx + i];
(gdb) bt
#0  0x00007ffff7b485a4 in depth_clustering::BytesTo<unsigned short> (data=..., start_idx=2835384544)
    at /home/tuan/workshop_ws/src/depth_clustering/src/ros_bridge/cloud_odom_ros_subscriber.cpp:45
#1  0x00007ffff7b42e23 in depth_clustering::CloudOdomRosSubscriber::RosCloudToCloud (this=0x7fffffffa610, msg=...)
    at /home/tuan/workshop_ws/src/depth_clustering/src/ros_bridge/cloud_odom_ros_subscriber.cpp:150
#2  0x00007ffff7b42a91 in depth_clustering::CloudOdomRosSubscriber::CallbackVelodyne (this=0x7fffffffa610, 
    msg_cloud=...) at /home/tuan/workshop_ws/src/depth_clustering/src/ros_bridge/cloud_odom_ros_subscriber.cpp:121
#3  0x00007ffff7b64d00 in boost::_mfi::mf1<void, depth_clustering::CloudOdomRosSubscriber, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&>::operator() (this=0xd4b078, p=0x7fffffffa610, a1=...)
    at /usr/include/boost/bind/mem_fn_template.hpp:165
#4  0x00007ffff7b60624 in boost::_bi::list2<boost::_bi::value<depth_clustering::CloudOdomRosSubscriber*>, boost::arg<1> >::operator()<boost::_mfi::mf1<void, depth_clustering::CloudOdomRosSubscriber, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&>, boost::_bi::list1<boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&> > (this=0xd4b088, f=..., a=...) at /usr/include/boost/bind/bind.hpp:313
#5  0x00007ffff7b5b961 in boost::_bi::bind_t<void, boost::_mfi::mf1<void, depth_clustering::CloudOdomRosSubscriber, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&>, boost::_bi::list2<boost::_bi::value<depth_clustering::CloudOdomRosSubscriber*>, boost::arg<1> > >::operator()<boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&> (this=0xd4b078, a1=...) at /usr/include/boost/bind/bind.hpp:905
#6  0x00007ffff7b5737e in boost::detail::function::void_function_obj_invoker1<boost::_bi::bind_t<void, boost::_mfi::mf1<void, depth_clustering::CloudOdomRosSubscriber, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&>, boost::_bi::list2<boost::_bi::value<depth_clustering::CloudOdomRosSubscriber*>, boost::arg<1> > >, void, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&>::invoke (
    function_obj_ptr=..., a0=...) at /usr/include/boost/function/function_template.hpp:159
#7  0x00007ffff7b607b5 in boost::function1<void, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&>::operator() (this=0xd4b070, a0=...) at /usr/include/boost/function/function_template.hpp:773
#8  0x00007ffff7b5bad9 in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const>) (function_obj_ptr=..., a0=...)
    at /usr/include/boost/function/function_template.hpp:159
#9  0x00007ffff7b90a6e in boost::function1<void, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> >::operator() (this=0xd4fe98, a0=...) at /usr/include/boost/function/function_template.hpp:773
#10 0x00007ffff7b8f583 in message_filters::CallbackHelper1T<boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&, sensor_msgs::PointCloud2_<std::allocator<void> > >::call (this=0xd4fe90, event=..., 
    nonconst_force_copy=false) at /opt/ros/kinetic/include/message_filters/signal1.h:76
#11 0x00007ffff7b54741 in message_filters::Signal1<sensor_msgs::PointCloud2_<std::allocator<void> > >::call (
    this=0xd4f838, event=...) at /opt/ros/kinetic/include/message_filters/signal1.h:119
---Type <return> to continue, or q <return> to quit---
#12 0x00007ffff7b50a0b in message_filters::SimpleFilter<sensor_msgs::PointCloud2_<std::allocator<void> > >::signalMessage (this=0xd4f838, event=...) at /opt/ros/kinetic/include/message_filters/simple_filter.h:136
#13 0x00007ffff7b4d507 in message_filters::Subscriber<sensor_msgs::PointCloud2_<std::allocator<void> > >::cb (
    this=0xd4f830, e=...) at /opt/ros/kinetic/include/message_filters/subscriber.h:206
#14 0x00007ffff7b6589c in boost::_mfi::mf1<void, message_filters::Subscriber<sensor_msgs::PointCloud2_<std::allocator<void> > >, ros::MessageEvent<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&>::operator() (
    this=0xd4f7f0, p=0xd4f830, a1=...) at /usr/include/boost/bind/mem_fn_template.hpp:165
#15 0x00007ffff7b6126e in boost::_bi::list2<boost::_bi::value<message_filters::Subscriber<sensor_msgs::PointCloud2_<std::allocator<void> > >*>, boost::arg<1> >::operator()<boost::_mfi::mf1<void, message_filters::Subscriber<sensor_msgs::PointCloud2_<std::allocator<void> > >, ros::MessageEvent<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&>, boost::_bi::list1<ros::MessageEvent<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&> >
    (this=0xd4f800, f=..., a=...) at /usr/include/boost/bind/bind.hpp:313
#16 0x00007ffff7b5cb19 in boost::_bi::bind_t<void, boost::_mfi::mf1<void, message_filters::Subscriber<sensor_msgs::PointCloud2_<std::allocator<void> > >, ros::MessageEvent<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&>, boost::_bi::list2<boost::_bi::value<message_filters::Subscriber<sensor_msgs::PointCloud2_<std::allocator<void> > >*>, boost::arg<1> > >::operator()<ros::MessageEvent<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&> (this=0xd4f7f0, a1=...) at /usr/include/boost/bind/bind.hpp:905
#17 0x00007ffff7b5892f in boost::detail::function::void_function_obj_invoker1<boost::_bi::bind_t<void, boost::_mfi::mf1<void, message_filters::Subscriber<sensor_msgs::PointCloud2_<std::allocator<void> > >, ros::MessageEvent<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&>, boost::_bi::list2<boost::_bi::value<message_filters::Subscriber<sensor_msgs::PointCloud2_<std::allocator<void> > >*>, boost::arg<1> > >, void, ros::MessageEvent<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&>::invoke (function_obj_ptr=..., a0=...)
    at /usr/include/boost/function/function_template.hpp:159
#18 0x00007ffff7b8fc5b in boost::function1<void, ros::MessageEvent<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&>::operator() (this=0xd4f7e8, a0=...) at /usr/include/boost/function/function_template.hpp:773
#19 0x00007ffff7b8f461 in ros::SubscriptionCallbackHelperT<ros::MessageEvent<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&, void>::call (this=0xd4f7e0, params=...)
    at /opt/ros/kinetic/include/ros/subscription_callback_helper.h:144
#20 0x00007ffff6d64e2d in ros::SubscriptionQueue::call() () from /opt/ros/kinetic/lib/libroscpp.so
#21 0x00007ffff6d0a6f8 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) ()
   from /opt/ros/kinetic/lib/libroscpp.so
#22 0x00007ffff6d0c0fb in ros::CallbackQueue::callAvailable(ros::WallDuration) ()
   from /opt/ros/kinetic/lib/libroscpp.so
#23 0x00007ffff6d68344 in ros::AsyncSpinnerImpl::threadFunc() () from /opt/ros/kinetic/lib/libroscpp.so
#24 0x00007fffefa2f5d5 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
#25 0x00007ffff71c86ba in start_thread () from /lib/x86_64-linux-gnu/libpthread.so.0

```
Do you have any idea how to fix this?
Thanks!

tuandle, 2018-11-21:
@niosus any help, please?

niosus, 2018-11-21:
Hello, sorry, missed the issue here. I have not tested it on rosbag file from kitti, but I assume that this is related to other issues opened here, #14 #5 #18 

Make sure the data is in the format the code expects here. Some data might be missing from the `PointCloud2` message here.

tuandle, 2018-11-21:
@niosus Thanks for your tip. You are right. The pointcloud messages in the rosbag don't contain the ring field.

closed 2018-11-21 by tuandle