PRBonn/depth_clustering #31: Eigen fatal when runing

LiShuaixin, 2018-11-27:
Hi, there. I try to show objects using moosmann dataset but failed.

The running environment is ubuntu 16.04 with ros kinetic and eigen 3.3.4. 

I got some feedback as follow:

    INFO: Reading from: /home/lee/depth_clustering_ws/data/scenario1/
    INFO: running on Moosman data
    INFO: Getting file paths from folder: /home/lee/depth_clustering_ws/data/scenario1/
    INFO: There are 2513 '.png' files in the folder.
    INFO: Getting file paths from folder: /home/lee/depth_clustering_ws/data/scenario1/
    INFO: There are 1 'img.cfg' files in the folder.
    INFO: Set en_US.UTF-8 locale.
    INFO: Reading config.
    INFO: Skipping commentary:
    # imgHSize; imgVSize; horizStartAngle(grad); horizStopAngle(grad); vertAngle_1; vertAngle_2; ...; vertAngle_n
    start:180.000000, stop:-180.000000, span:360.000000, step:-0.413793
    INFO: Params sucessfully read. Rows: 64, Cols: 870

    ===================== Setting Connection =====================
    || Sender: DepthGroundRemover (id: 3)
    || Type: STREAMER
    || |
    || V
    || Client: ImageBasedClustere<(short)1, (short)1> > (id: 5)
    ==============================================================

    ===================== Setting Connection =====================
    || Sender: ImageBasedClustere<(short)1, (short)1> > (id: 5)
    || Type: STREAMER
    || |
    || V
    || Client: ObjectPtrStorer (id: 2)
    ==============================================================

    show_objects_moosmann: /usr/local/include/eigen3/Eigen/src/Core/DenseStorage.h:128：Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = float; int Size = 16; int MatrixOrArrayOptions = 0]: Assertion ‘(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"’ Failed。

I checked the web and found it's the problem caused by Eigen. Is there anyone know how to fix it? Many thanks!

niosus, 2018-11-27:
That might mean that I am using an unaligned `std::vector` with Eigen contents somewhere in my code. I will try to look for it.

LiShuaixin, 2018-11-27:
> That might mean that I am using an unaligned `std::vector` with Eigen contents somewhere in my code. I will try to look for it.

I checked the code, and found the error occurred in line 73 in show_objects_moosmann.cpp. 

I notice that there is a member std::vector<RichPoint> _points, while the class RichPoint having Eigen::Vector member. It seems like the [cause 1](http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html) expressed in the web.

niosus, 2018-11-27:
Thanks, I have done s small PR that should fix this. Thanks for pointing out this issue, it is clearly a bug. If the tests and build pass, I will merge it today.

LiShuaixin, 2018-11-27:
> Thanks, I have done s small PR that should fix this. Thanks for pointing out this issue, it is clearly a bug. If the tests and build pass, I will merge it today.

Hi, @niosus, thanks for your reply and help, I try to run the fixed code, but the problem is still not solved. Could you please check it once again?

niosus, 2018-11-27:
Are you sure you have rebuilt everything? That was the only place in the code where I used `std::vector` with Eigen types.

LiShuaixin, 2018-11-27:
I'm sure about that. I even try to clone the latest code once again and rebuild it using cmake build as the instruction in README.

niosus, 2018-11-27:
Weird. I have ran the code against an address sanitizer and there are no false reads in the code on my machines... Can you try to add `EIGEN_MAKE_ALIGNED_OPERATOR_NEW` right under `class Cloud {` line in the `src/utils/cloud.h` and see if this fixes your issue?

LiShuaixin, 2018-11-28:
Hi, @niosus, I used another PC with the same environment (ubuntu16.04 + Eigen 3.3.4) could run the code smoothly. It's so weird! It should be a individual case for me now instead of a bug in the code, so you could close the issue. Thanks for your devotion to this work!

LiShuaixin, 2018-11-28:
Hi, @niosus, I change to use another PC with the same environment (ubuntu16.04 + Eigen 3.3.4) which could run the code smoothly. It's so weird! Obviously, it should be a individual case for me now instead of a bug in the code, so you could close the issue. Thanks for your devotion to this work!

niosus, 2018-11-28:
Usually if it crashes on one machine there is a bug somewhere. Anyway, it's gonna be hard to debug, so if the code runs well for you now, I will close this issue.

PR #32 (2018-11-27): Fix Eigen point alignment in a vector
I was using the unaligned vector of Eigen points, which is clearly a bug. This PR fixes this. 

Fix #31
closed 2018-11-28 by niosus