PRBonn/depth_clustering #33: I have correted vlp32 with height:32 and width:2172, but still got "Not correcting depth data"

kxhit, 2018-12-12:
Thanks for your work!
I tried to publish vlp-32 points with data size of 32x2172 (This is structured data), but still got this warning "Not correcting depth data". I think the reason is that the data size is incorrect. What is the correct size of structured data of vlp32 and where can I change it? 
Thanks! Looking forward to your reply.

niosus, 2018-12-12:
This is fine and should not be a warning. It is supposed to be a debug message and the corrections are just applied in the case of 64 beam velodyne where I saw the calibration data to be suboptimal.

closed 2018-12-12 by niosus

hunkyu, 2019-02-28:
I added the vlp32c parameter, but the result doesn't feel right. How do you set it up?
std::unique_ptr<ProjectionParams> ProjectionParams::VLP_32_C()
{
    auto params = ProjectionParams();
    params.SetSpan(SpanParams(-180_deg, 180_deg, 870),
                   SpanParams::Direction::HORIZONTAL);
    params.SetSpan(SpanParams(2.33_deg, 25_deg, 32),
                   SpanParams::Direction::VERTICAL);
    params.FillCosSin();
    if (!params.valid()) {
      fprintf(stderr, "ERROR: params are not valid!\n");
      return nullptr;
    }
    return mem_utils::make_unique<ProjectionParams>(params);
}
[image]

niosus, 2019-02-28:
Hey @hunkyu 
You should not worry about that message. It is not an error. It is just a debug message for me. The correction is only needed when running with kitti data as it has had internal lidar calibration.