PRBonn/depth_clustering #34: Why do we need to fix kitti depth

yiakwy, 2019-01-09:
Hi, I am new to kitti and I am reading your code to compare it with using  PCL to segment points cloud.  I noticed that in your code, we need to fix every picture from "moosman" dataset, any special reason may I know?

niosus, 2019-01-09:
If you look at the endpoints of the beams on a straight wall you will see that the endpoints are not on the same plane, which indicates some issues with sensor calibration (or with the parameters that I was using). This has a huge influence on the angles I estimate in my code and therefore I decided to fix this by approximating the changes that need to be added to the beams to account for this.

closed 2019-01-09 by niosus

yiakwy, 2019-02-11:
@niosus Thank you for your time and reply.