PRBonn/depth_clustering #36: How to get the point clouds of road ground and publish them as a ros topic?

firesnow1234, 2019-01-10:
Hi
Thanks for your sharing!
You have removed ground points in the range image, how can we get the point clouds of road ground and publish them?  and how to publish bounding box of objects as a ros topic？

niosus, 2019-01-14:
Hello! I do not publish ROS topics as I only use this code within my own pipeline and I do not communicate through ROS. But feel free to check out other issues for discussions on that. I think people have started some forks to make this functionality available.

closed 2019-01-14 by niosus

firesnow1234, 2019-01-16:
So is there any way to get the passable road area?

niosus, 2019-01-16:
What do you mean by "passable road data" here? If you want to find the road in the part of the cloud that corresponds to ground, you will have to do it yourself. The code here just separates the ground from everything else.