PRBonn/depth_clustering #38: Run ./show_objects_node fail

phamquandung, 2019-01-28:
I tried to run ./show_objects_node by following this link https://wiki.nps.edu/pages/viewpage.action?pageId=925958215 (I convert the kitti raw data to bag) but the result is nothing. 
This is the process I tried to do:
First step:
[image]
Second step:
[image]
Third step:
[image]
And the result is:
[image]

Could you please help me? I really don't understand what's wrong with it because I follow the guide above.

niosus, 2019-01-29:
I'm sorry, I won't have time to fully debug this in the near future. :( 
However, the first idea would be to check if the data is published under the topic that my code expects.

phamquandung, 2019-02-08:
Dear mate. I was wondering if your code has evaluation part? I mean did you code the evaluation which compares to Euclidean and grid-based? Moreover, in your paper, I did not see the source of ground truth dataset you use to compare. Could you please tell me the source of ground truth? Thank you!

niosus, 2019-02-23:
@phamquandung please don't start new threads inside other threads. If you have a new question, just open a new issue. The evaluation was relying on a multitude of scripts and has never matured to be open sourced unfortunately. I will not have the time to prepare the data and make is available unfortunately, sorry about that.

closed 2019-02-23 by niosus