PRBonn/depth_clustering #40: how to get depth image

Chrislzy1993, 2019-02-25:
hi, @niosus, thanks for your code. i turn pointcloud into depth image , but the result is different from `Frank Moosmann's "Velodyne SLAM" Dataset`, and i see the image size is 64 X 870, i guss there maybe some fileter process in the code , and i want to know how to do this???? thanks again

Zhang-Qinghui, 2019-10-20:
> hi, @niosus, thanks for your code. i turn pointcloud into depth image , but the result is different from `Frank Moosmann's "Velodyne SLAM" Dataset`, and i see the image size is 64 X 870, i guss there maybe some fileter process in the code , and i want to know how to do this???? thanks again

hello，Chrislzy1993. I also want to turn piontcloud into depth image, but i don't know how to do this. which file in the project did this thing? thanks.

niosus, 2019-11-21:
@Chrislzy1993 I'm not really getting the issue. In the Frank Moosmann's that you can get following the link in the readme, all the images are 64x870. Mind that this is not the full resolution of an HDL64 Velodyne, but seems to be a good enough simplification that works quite well in practice.

closed 2019-11-21 by niosus