PRBonn/depth_clustering #45: kernel problem

lonlonago, 2019-06-20:
1    kernel = Mat::zeros(window_size, 1, CV_32F);
      kernel.at<float>(0, 0) = -3.0f;
      kernel.at<float>(0, 1) = 12.0f;
      kernel.at<float>(0, 2) = 17.0f;
      kernel.at<float>(0, 3) = 12.0f;
      kernel.at<float>(0, 4) = -3.0f;
      kernel /= 35.0f;
      return kernel;

the kernel size first is **window_size*1**, why set value (0, 1) (0, 2)...  does kernel size is **1*window_size**

2 in  GetUniformKernel
  Mat kernel = Mat::zeros(window_size, 1, type);
  kernel.at<float>(0, 0) = 1;
  kernel.at<float>(window_size - 1, 0) = 1;
  kernel /= 2;

kernel /= 2 make kernel is like [[0.5, 0, 0.... 0,  0.5]] , but kernel value  must be 1, why devidi by 2?

niosus, 2019-06-20:
Sorry that the code is not expressive enough. :) 
The uniform kernel is not really a uniform one. It is basically an averaging kernel that skips over some values. For the Savitsky Golay one please look up the relevant paper where they explain their rationale for picking those values. I hope that answers your question.

lonlonago, 2019-06-21:
@niosus  thanks for your reply!
For the Savitsky Golay , I am not question about the value ,but the size, you should use  **kernel.at<float>(3,0) = 12.0f;** not **kernel.at<float>(0, 3) = 12.0f;**

For the uniform kernel,you use it in **cv::dilate**, like **cv::dilate(*label_image_ptr, dilated, kernel)**, but cv::dilate kernel parameter need to be 0 or 1, not like 0.5 value.

thanks again!

niosus, 2019-06-21:
Oh, you might be right. I will have a look at this. I always mix up rows and cols. Now I wonder how it works in the first place :)

MikeZhu92, 2019-07-12:
@niosus I've verified the kernel value and it's correct. Since it's a 1d matrix mixing up the row col doesn't have big impact.

However, I'm not sure about the dimension of the kernel.
Right now the kernel is `kernel = Mat::zeros(window_size, 1, CV_32F);` so it's `window_size x 1`.
Since the angle image size is 64 x number of scans (using HDL 64 as example), it might make more sense to apply the kernel row by row from left  to right.

niosus, 2019-11-21:
The kernel is only applied to a single column at a time, so that checks out for me. Or did I get your issue wrong?

niosus, 2020-01-10:
Closing for lack of activity.

closed 2020-01-10 by niosus