PRBonn/depth_clustering #52: Really bad visualization

yrt1993, 2019-10-24:
I don't know why the author show the segmentation result like this. When I run examples in the codes like show_objects_kitti and qt_gui_app, I see the original point cloud of white color with very many yellow or blue 3d bounding boxes and even more grey 3d bounding boxes. 

What I want to see is all those segmented object as well as the ground drawn with different colors and bounding boxes are dispensable. Then I get a visualized qualitative evaluation of the segmentation algorithm implemented in the depth-clustering 
[image]

niosus, 2019-11-21:
The way your data is visualized looks weird. The vis in the readme is from the qt visualization, see the binaries provided. This will enable you to play with the parameters and hopefully improve the performance.

niosus, 2020-01-10:
Closing for the lack of activity.

closed 2020-01-10 by niosus