PRBonn/depth_clustering #62: Unable to build

bymbhaskar, 2020-02-12:
I am unable to build using ROS Kinetic (ubuntu 16.04).
Showing following error:
 error: call of overloaded ‘make_shared<PclCloud>(PclCloud&)’ is ambiguous
   return make_shared<PclCloud>(pcl_cloud);
                                         ^
[image]

niosus, 2020-02-12:
Oh, strange. Could you modify this code with `boost::make_shared` instead of a simple `make_shared` and see if this helps? Strange that it builds on Ubuntu 18.04 though...

bymbhaskar, 2020-02-12:
Thank you for your quick response. 
I tried with boost::make_shared. It still didn't work..
error is as follows:  

/home/bhaskar/catkin_ws/devel/.private/depth_clustering/lib/libfolder_reader.so: undefined reference to `boost::re_detail::cpp_regex_traits_implementation<char>::transform[abi:cxx11](char const*, char const*) const'
/home/bhaskar/catkin_ws/devel/.private/depth_clustering/lib/libfolder_reader.so: undefined reference to `boost::re_detail::cpp_regex_traits_implementation<char>::transform_primary[abi:cxx11](char const*, char const*) const'
collect2: error: ld returned 1 exit status

[image]

bymbhaskar, 2020-02-13:
able to build now.
Thanks

closed 2020-02-13 by bymbhaskar

niosus, 2020-02-13:
Could you point out the changes you had to make?

bymbhaskar, 2020-03-07:
I just went for Ubuntu 18.04 and was able to build

niosus, 2020-03-07:
Thanks for the update. It's still weird that you could not build on other versions of Ubuntu though...

DavidZhaoP, 2022-02-13:
> > Oh, strange. Could you modify this code with `boost::make_shared` instead of a simple `make_shared` and see if this helps? Strange that it builds on Ubuntu 18.04 though...
> 
> I met the same problem, and i follow your solution by ,modify this code with boost::make_shared instead of a simple make_shared. But another problem happened,like this [image]
> 
> Can you give me some help? Thanks
 
problem solved!
use std::make_shared instead of using boost::make_shared .