PRBonn/depth_clustering #64: Implementation on velodyne lidar

bymbhaskar, 2020-03-07:
I am using Velodyne (VLP-32C) Lidar. Could you please let me know, if this Lidar gives range data (as 32 rows and many columns, used in your method) or I need to create this kind of format using the point cloud. Creating a cylindrical projection from a point cloud increases the run time to great extent..

niosus, 2020-03-08:
If you have your own driver for a lidar, you can definitely implement it in a way where it gives you the image (or rather multiple small "scan" images). If you don't own a driver, it depends on what it provides you. Projecting the data takes really not that long even if you only have a point cloud. Definitely enough for prototyping, and if you actually want to use this in production, you should probably have your own driver.

bymbhaskar, 2020-03-09:
We have Velodyne LiDAR sensor but can you please clarify, what exactly you mean by "driver"?

niosus, 2020-03-09:
The driver is the software that interprets the data coming directly from the sensor and creates a useful representation that is used further down the line, i.e., to point clouds.

bymbhaskar, 2020-03-10:
Thanks for ur reply. I am getting a run time of approx. 0.02 second for each frame on 32GB RAM system. Is this run time justified or could there be a problem in implementation , in ur opinion?

niosus, 2020-03-11:
The amount of RAM is relatively irrelevant here. The crucial factor is what is your processor. Otherwise, you are getting around 50fps, which seems rather slow. What are you measuring?

I will close this issue, but feel free to continue the discussion here.

closed 2020-03-11 by niosus