PRBonn/depth_clustering #69: Release for ROS Noetic

UniBwTAS, 2020-06-29:
Will there be a release for ROS Noetic? Thank you for your great work.

niosus, 2020-07-09:
I am happy to merge a PR about this, but probably won't have the time to port it myself.

UniBwTAS, 2020-07-10:
We will work on this soon.

zw199502, 2023-01-25:
I successfully released the project for ROS Noetic. Please follow the following:

1. ubuntu20.04 + ros-noetic

2. sudo apt install libqglviewer-dev-qt5 freeglut3-dev qtbase5-dev 

3. the default opencv version is 3, please revise the related opencv codes same as the format of opencv4 
#include <opencv2/highgui/highgui.hpp> -> #include<opencv2/highgui/highgui_c.h> 
#include <opencv/cv.h> -> #include <opencv2/imgproc.hpp> 
#include <opencv/highgui.h> -> #include<opencv2/highgui/highgui_c.h> 
CV_LOAD_IMAGE_ANYDEPTH -> cv::IMREAD_ANYDEPTH

CMakeLists.txt
set(CMAKE_CXX_STANDARD 11) -> set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_FLAGS "-Wall -Wextra -fPIC") -> set(CMAKE_CXX_FLAGS "-Wall -Wextra -fPIC -pthread")

cloud.cpp
return make_shared<PclCloud>(pcl_cloud); -> return boost::make_shared<PclCloud>(pcl_cloud);

4. catkin build depth_clustering -DPYTHON_EXECUTABLE=/usr/bin/python3 
you should install catkin build tools by yourself. 
Please do not build in Anaconda environment becasue of the conflict of Qt version

SunZezhou, 2023-05-08:
Thanks to the steps above, for me one additional step needs to be added for it to work properly:

CMakeLists.txt  line 93

elseif(${DISTRO} MATCHES "16.04" OR ${DISTRO} MATCHES "18.04") -> elseif(${DISTRO} MATCHES "16.04" OR ${DISTRO} MATCHES "18.04" OR ${DISTRO} MATCHES "20.04")