PRBonn/depth_clustering #71: Query on Cluster Visualization

vin-cloud, 2021-03-04:
The visualization module within the code draws the cube for the detected clusters, but the .gif as seen below is visualized in the form of polygons! Was wondering if the polygon representation is included in the uploaded code!
Thank you so much!
[image]

niosus, 2021-03-04:
@vin-cloud the whole visualization pipeline could see some improvements, I would write it better now :wink: But now there is no more time for that. What you are looking for lives here https://github.com/PRBonn/depth_clustering/blob/fe9d6675b213926815bf2bbde593b28b55be4b98/src/qt/drawables/object_painter.cpp#L77-L101 

I just quickly implementer an OpenCV-based convex hull computation to improve the visualization.

niosus, 2021-03-08:
@vin-cloud I will close this issue for now. If you'd rather leave it open, feel free to reopen again.

closed 2021-03-08 by niosus