PRBonn/depth_clustering #75: build error

lejk8104, 2021-06-03:
some error when I build

Errors     << depth_clustering:make /home/dgist/workspace/git-folder/clustering_ws/logs/depth_clustering/build.make.002.log
CMakeFiles/depth_clustering_test.dir/test_pose.cpp.o: In function `PoseDeathTest_TestLikelihood_Test::TestBody()':
test_pose.cpp:(.text+0x585d): undefined reference to `testing::internal::DeathTest::Create(char const*, testing::Matcher<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>, char const*, int, testing::internal::DeathTest**)'
test_pose.cpp:(.text+0x5928): undefined reference to `testing::internal::DeathTest::Create(char const*, testing::Matcher<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>, char const*, int, testing::internal::DeathTest**)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/dgist/workspace/git-folder/clustering_ws/devel/.private/depth_clustering/lib/depth_clustering/depth_clustering_test] Error 1
make[1]: *** [test/CMakeFiles/depth_clustering_test.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
cd /home/dgist/workspace/git-folder/clustering_ws/build/depth_clustering; catkin build --get-env depth_clustering | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << depth_clustering:make                [ Exited with code 2 ]      
Failed    <<< depth_clustering 

what is happen?

my PC setting is
1. Ubuntu 18.04
2. ROS melodic
3. install PCL

my command is
$ pip install catkin-tools
$ catkin build depth_clustering

tuxiangxianzhe, 2021-11-11:
I have the same question, have you solved it?

tuxiangxianzhe, 2021-11-11:
I change the version of googletest to 1.7.0, and solved this problem.
I wish it helps you.

niosus, 2021-11-11:
@tuxiangxianzhe just for me to understand a bit better. Are you saying that you have changed the GoogleTest library version installed on your system? Which one is installed by default?

tuxiangxianzhe, 2021-11-11:
Since ubuntu 18.04 does not come with googletest, I previously install the version 1.11.0 from GitHub (https://github.com/google/googletest/releases), which is the latest version.

niosus, 2021-11-11:
Understood. I'll add that to the readme.

niosus, 2021-11-11:
I added the notice to the readme and am closing this issue.

closed 2021-11-11 by niosus

tuxiangxianzhe, 2021-11-11:
Un, maybe I am not clear about what I said.what I mean is that the latest version of Googletest will cause errors, and you need to install version 1.7.0 to work smoothly.