PRBonn/depth_clustering #77: running in real time

urbansound8K, 2022-01-04:
is it possible to run the code in real time data without using ROS and on live feed from cameras and lidar without saving them into png and bin format? in short what to do to run the lidar and feed the data into the network in real time !?

niosus, 2022-01-04:
It is definitely possible, but it needs to be implemented from scratch. You would need to look at the paper and reimplement the input into the algorithm. I will close this issue, but feel free to continue the discussion.

closed 2022-01-04 by niosus

urbansound8K, 2022-01-05:
I will go through that soon
but honestly, I am trying to skip ROS as I am learning it rn now and need the program to be implemented in a maritime environment. I have a question, please! did you run your program using ROS in real-time?

niosus, 2022-01-05:
The code here requires no ROS. There is no internal communication through ROS whatsoever. There is a single class to subscribe to ROS data which can be skipped. It is not used when the code is run on images. In your case, you will need to adapt the code to run on image-like structures that you can produce on your own depending on where your data comes from.